rurza/BatFi #1: Show watts ?

schabrolles, 2023-06-28:
Very nice App !!! 
Works perfectly.

Do you think you could add the current Watts used by the MAC ?
This could be interesting to have it when we click on the icon. What do you think ? 
[image]

I currently use a second app for that, but I think it is good to have everything in your app.
The best of the best would be to have :
  - Macbook current watt usage when disconnected 
  - Macbook current watt received when charging

rurza, 2023-07-04:
Hey! Not a bad idea :)
Do you know what API this app uses?:)

schabrolles, 2023-07-05:
I really don't know :) 
I use this App that report battery status : https://apps.apple.com/app/id836505650?mt=12

hiroshi80, 2023-07-06:
I'm constantly checking for apps that are consuming too much power⚡️.
So, I would be happy if the BatFi would notify me when the power consumption exceeds the threshold.

rurza, 2023-09-26:
I will add it with 1.4.0 :) There is a working version on the develop branch (thanks to @decodism), but I would like to polish the UI and I noticed that it once stopped working. I have to double check it before the release :)

closed 2024-01-17 by rurza