rurza/BatFi #9: Add "Sailing" mode

olfway, 2023-09-13:
This feature allows you to set upper and lower charge limit: do not charge while current value between lower and upper limits

Similar to this
https://apphousekitchen.com/feature-explanation-sailing-mode/

rurza, 2023-09-26:
Hey!
> Since it is theoretically healthier for the battery to be charged 10% once instead of 1% 10 times, we have included the Sailing Mode feature. This allows you to set a lower limit, from which the battery will be charged up to the charge limit again.

And then later:

> Which interval should I choose for my battery to last as long as possible?
Unfortunately, we could not find any explicit studies on this topic yet. Therefore, we assume a value of 5-10%, depending on your own judgment.

I will have to double check if it's actually needed.

olfway, 2023-09-26:
The idea is that a user can set any lower and upper limit as he want.
So, someone can set 45% – 65%, other user could set 90% – 95%

rurza, 2023-09-26:
I understand what it does and it seems like a really clever idea. I'm just curious if it actually has a noticeable effect on the lifespan :) 
And my data from the past 2 months doesn't support the claim that it happens frequently (it hasn't happened once on my M2 Max) – maybe because I don't leave my computer connected to the charger for long enough, or the Apple Silicon SOC doesn't consume enough power (maybe it happens on Intel), or perhaps I'm using chargers that are powerful enough (140W for M2 Max and 96W for M1 Pro), or maybe on Apple Silicon it only happens with the maxed version of the M Max. But please understand where I'm coming from – implementing this feature isn't a priority because another app has it, yet they don't provide any evidence that it actually affects the lifespan...

SohamJa1n, 2024-01-20:
> I understand what it does and it seems like a really clever idea. I'm just curious if it actually has a noticeable effect on the lifespan :)
> 
> > However, small discharges of the battery still occur when a lot of power is needed for a short time. Then the battery of the MacBook steps in as a buffer.
> 
> And my data from the past 2 months doesn't support the claim that it happens frequently (it hasn't happened once on my M2 Max) – maybe because I don't leave my computer connected to the charger for long enough, or the Apple Silicon SOC doesn't consume enough power (maybe it happens on Intel), or perhaps I'm using chargers that are powerful enough (140W for M2 Max and 96W for M1 Pro), or maybe on Apple Silicon it only happens with the maxed version of the M Max. But please understand where I'm coming from – implementing this feature isn't a priority because another app has it, yet they don't provide any evidence that it actually affects the lifespan...

Just installed this app and its great, love the statistics. I moved from another battery charge limiter to here, I did notice a change in battery health on my M2 macbook when sailing mode is enabled vs when not.

I understand it isn't a priority however just letting you know that it does make a difference.