rurza/BatFi #24: App with high energy usage doesn't show anything?

ectv96, 2024-01-17:
I have turned on the option, and lower to the lowest threshold, and still showing nothing.
on the macOS battery dock, it shows several apps that has high energy usage.

Macbook Air M2

rurza, 2024-01-17:
Hm... I wasn't using this functionality for some time. It uses private APIs behind the scene so there are 2 options:
- regression
- macOS update broke it 

I will take a look at it today :)

rurza, 2024-01-17:
@ectv96 
Hello! After double-checking on my personal machine, I couldn't reproduce the issue. I will continue to monitor it. Could a restart of your Mac help? Also, is your machine managed by IT (e.g., through MDM or similar)?

squishycat92, 2024-01-18:
The reason for this is probably because the lowest threshold for BatFi (energy impact of 200) is significantly lower than the default macOS Battery menu's threshold. I'm not sure how macOS determines whether or not an app is using significant energy, but for example, when Safari is using 87.1, it shows up in the stock Battery menu. However, since the impact is <200, it does not show in BatFi.

Maybe it would be possible to lower the threshold further to whatever macOS considers using significant energy?

rurza, 2024-01-18:
Absolutely, that's not it, but I appreciate the suggestion. :) 
On my work machine, even when I compile a rather sizable Swift project (one that takes a few minutes on the M2 Max), the list is empty. For now, my strategy is to enhance the logging in the latest beta, and I aim to deduce the issue solely from the logs. 
Additionally, on my private M3 Max where I can't reproduce the problem, compiling NepTunes.app almost instantly reveals the spike.

[image]

rurza, 2024-01-18:
I mean I can lower the threshold, but it won't solve the problem – apparently sometimes it just doesn't work at all :(

rurza, 2024-01-19:
@decodism Lol, yes :)

[image]