rurza/BatFi #30: Update to Version 1.4.1 (32) has all the language set to cyrillic characters

rna84, 2024-02-04:
I've updated today to the newer version but now all the text is set to cyrillic characters instead of English (or latin characters)
[image]
[image]

rurza, 2024-02-08:
Hey @rna84 :)
What language do you have set in System Settings? (Open System Settings -> General -> Language & Region)

rna84, 2024-02-08:
I think I've fixed it, I had Russian set on it - now by setting another language is everything fine. Thanks!

closed 2024-02-08 by rna84