rurza/BatFi #34: Incorrect battery health

Neifab, 2024-03-12:
Hello there,

I've noticed an incorrect battery health pourcentage in the app. CoconutBattery tells me it's at 95,9 and BatFi at 100%.

I guess BatFi is taking the "official" macOS battery statut, maybe there is a way to put another one more accurate.

Best

rurza, 2024-03-14:
Hey! 
When the app was displaying calculated capacity, I was receiving a bunch of emails every week about it showing the "wrong" health. I'm not sure which one is actually "right," but because the OS consistently reports 100% for a new battery (while CoconutBattery might report values like 101% and then 95% two weeks later), I've decided to use the one provided by the OS.

closed 2024-03-14 by rurza

Neifab, 2024-03-23:
Thanks for the answer