rurza/BatFi #36: Change notification %

vardecab, 2024-03-19:
Looks like a great app! Thanks for building and releasing it.

Would it be possible to change the notification from 20% to user selected %? 

Use case: user wants to keep battery between 30-70% so sets up charging limit to 70% but notification is currently fixed at "low battery 20%" so user needs to check manually if the battery is at 30% or not.

Ideally user can set charging limit % AND low battery % like in [Battery Monitor](https://apps.apple.com/us/app/battery-monitor-health-info/id836505650?mt=12):
[image]

rurza, 2024-03-21:
Hey!
Good idea :) Consider it done

finxol, 2024-11-20:
Pretty sure this has been addressed. From what I understand, UI is in [NotificationsView.swift](https://github.com/rurza/BatFi/blob/main/BatFiKit/Sources/Settings/NotificationsView.swift#L34).