rurza/BatFi #40: "Initializing" bug in 1.5.0

vardecab, 2024-04-22:
After updating to beta 1.5.0 (to get low battery notification at 30%! #36 :)) the app is stuck. I tried to uninstall, reinstall, change various settings, unplugged cable but nothing seems to work. It's just _Initializing_ and the laptop continues to charge ¿\_(ツ)_/¿

[image]

Going back to stable build helps. Updating from 1.4.1 to 1.5.0 messes things up again. Is there a direct download to 1.5.0 so I can check clean install vs updating from 1.4.x?

jainal09, 2024-04-22:
I am having the same issue in 1.4.2 as well

vardecab, 2024-04-22:
> I am having the same issue in 1.4.2 as well. How to find the commit for the 1.4.1 version @vardecab ?

I simply re-download from [here](https://micropixels.software/apps/batfi).

vardecab, 2024-04-23:
Hmmm, 1.4.1 stopped working as well.

rurza, 2024-04-25:
Hey!
Does the restart helps? 
Install the version you want (you don't have to launch it), restart and you should be good to go.
_Keep in mind that there has to be a single copy of the app on your disk_

vardecab, 2024-04-26:
Uninstalled everything, restarted, installed 1.4.1, restarted, updated to 1.5.0, restarted — now it works! No idea why it didn't before. 

Thanks @rurza :)

closed 2024-04-26 by vardecab