rurza/BatFi #53: "Automatically enable system charge limit when the Mac goes to sleep" gives me 3-5 seconds delay for waking up my screen after sleep

rmaprojects, 2024-06-07:
My Mac: Macbook Air M1 (Late 2020)

This happened after I enabled Automatically enable system charge limit when the Mac goes to sleep feature, I usually lock my mac when I was doing something else outside of my Mac, and let the screen dim and sleep by itself. Usually, the screen lightens up when I press the trackpad and touch ID. But now I have to wait for about 3-5 seconds for Mac to responding my clicks. After waiting for 3-5 seconds and unlocking the screen, everything back to normal and no delay anymore. But I hate those delays.

This issue happened when I left my mac sleeps when it is charging and sleeping.

Can you please take a look for the bug, please?

lestitaniseur, 2024-08-08:
Also same problem but I  figurate out the problem 

To reproduce the Bug you have the check auto system charge limit in sleep then turn your Mac to sleep then plug the charger then you have to cover the notch before the lid of your Mac turn 
when this feature is activated the Mac use the notch light sensor to walk up not when you open the lid this is very frustrating bug pls patch

lestitaniseur, 2024-08-08:
> My Mac: Macbook Air M1 (Late 2020)
>
> This happened after I enabled Automatically enable system charge limit when the Mac goes to sleep feature, I usually lock my mac when I was doing something else [...]

when this option is activated the Mac use the notch sensor to wake up I don't know why this is related to batfi

lestitaniseur, 2024-11-01:
Any update I figure something else this problem occurs when the option of Automaticallv adiust briahtness is off but when this option is on it fix the problem but when this feature is off the Mac use the notch to wake up