rurza/BatFi #60: Charge to 100% doesn't work

ishaanrathodd, 2024-06-27:
I have my charge limit set to 90% and it holds charging till 90% even if i enable charge to 100% option. Even if I turn off automatically manage charging, it doesn't go above 90%. Worst part, even if i completely uninstall the app, the charging still holds up to 90%, it never goes above the charge limit i set in the app. I think something is conflicting, the app has changed the system settings so it follows the charge limit set in the app even after i uninstall the app completely.

ishaanrathodd, 2024-07-18:
@rurza please look into this, its like theres no possible way to charge to 100% if i dont reinstall the OS

rurza, 2024-07-18:
And it doesn't charge even if you quit the app?

ishaanrathodd, 2024-07-19:
> And it doesn't charge even if you quit the app?

it doesnt charge even if i uninstall the app, it has changed the internal settings

rurza, 2024-07-19:
This clearly proves that BatFi isn’t the cause. You’ve installed something that interferes with power management.

closed 2024-07-19 by rurza

ishaanrathodd, 2024-07-19:
> This clearly proves that BatFi isn’t the cause. You’ve installed something that interferes with power management.

never installed anything other than batfi for battery related stuff, as soon as i installed batfi, im getting this problem though

rurza, 2024-07-19:
And it persists even after restarting Mac?

ishaanrathodd, 2024-07-19:
> And it persists even after restarting Mac?

yes, no way to fix it, tried everything, moreover the automatically manage charging also doesnt work, even if i turn it off, the battery is stuck at 80, optimized charging is turned off always so thats not the one causing problem is my guess

rurza, 2024-07-19:
BatFi doesn’t set the limit; it pauses charging. The exception is the 80% system charging limit set during sleep and turned off when awake. Additionally, upon quitting, it resets all settings, and after a reboot, macOS resets all settings.

ishaanrathodd, 2024-07-19:
> BatFi doesn’t set the limit; it pauses charging. The exception is the 80% system charging limit set during sleep and turned off when awake. Additionally, upon quitting, it resets all settings, and after a reboot, macOS resets all settings.

understandable, though seems weird why this happened only after batfi was installed. do u have any suggestions on what can i do to redo this?