rurza/BatFi #61: MacOS 15 Bug: Overlay of 0 and Battery Percentage

Xytronix, 2024-06-27:
In the latest macos 15 beta, batfi overlays 0 and the battery percentage.

After force closing batfi, I had to do the onboarding again.
I couldn't access the settings, instead **!** was shown as the battery percentage.
The buttons to settings and other pages were faded out. The menu that shows after clicking the batfi icon in the menu bar.
Bug was solved by removing batfi from being able to run in background and enabling it.

Xytronix, 2024-06-28:
Another bug, Batfi stops charging at 46% even if set to 80% in settings.
[image]
[image]

rurza, 2024-06-28:
Are you sure you don't have "Automatically turn off charging when the battery gets hot" turned on (in the Advanced pane)? It's on by default. Since you have the UI in German, I assume the recent heat wave could have an impact. 🙂

Xytronix, 2024-07-05:
> @Xytronix Are you sure you don't have "Automatically turn off charging when the battery gets hot" turned on (in the Advanced pane)? It's on by default. Since you have the UI in German, I assume the recent heat wave could have an impact. 🙂

@rurza could be it, but why did a reboot fixed the issue then?