rurza/BatFi #67: Menubar icon showing exclamation mark and Power Distribution always waiting for Data

RuochenFu21, 2024-07-09:
Like this: 
[image]
(Mac OS 15 Beta, most recent version of BatFi)

closed 2024-07-09 by rurza

RuochenFu21, 2024-07-10:
Uh how to solve this?

rurza, 2024-07-10:
Please follow up the troubleshooting section on app's website:

https://micropixels.software/apps/batfi#faq

RuochenFu21, 2024-07-11:
App mode stayed at Initializing after try to following the procedure under "I've set the limit, but the battery keeps charging". I couldn't find BaFi.app in "Allow in the Background" but found "BatFiHelper" and it was enabled. What to do now?

rurza, 2024-07-12:
If you're using macOS **DEVELOPER BETA** consider the following:
- always use the latest version before reporting an issue
- avoid using it if you’re not a macOS developer. It’s intended for developing apps, so it’s completely expected that third-party apps may not work properly

0PandaDEV, 2025-06-30:
I have the same issue, I'm on a fresh installation of macOS Sequoia.