rurza/BatFi #82: Can't inhibit charging on sleep mode

berkeenesakt, 2024-09-21:
After macOS Sequoia update app stopped inhibit charging on sleep mode. I enabled charge limit on sleep mode. 
[image]

manishdm123, 2024-09-21:
I have noticed the same issue

berkeenesakt, 2024-09-22:
It's really huge problem makes the application non-functional. We are waiting fix asap @rurza 🙏

Sin-cy, 2024-09-22:
Yeah its happened after the Mac OS Sequoia update.

rurza, 2024-09-22:
Hey folks!
Did you tried the latest BatFi Beta? :)

Sin-cy, 2024-09-23:
@rurza I just got an update for version 2.1.0. Read the patch notes. Thank you.

> Hey folks! Have you tried the latest BatFi Beta? :)

rmaprojects, 2024-09-23:
Now, inhibit charging on sleep mode is completely disappear anyone know how to turn it back?

[image]

berkeenesakt, 2024-09-24:
I think it's removed. 

Anyone using this version of macOS, or those who updated it on a separate volume, will lose this functionality, even when running macOS 13 or 14. The update hides the toggle switch in the app’s settings for macOS 15 users

berkeenesakt, 2024-09-24:
@rurza I saw on another app, there is another option for stop charging when Mac goes sleep instead of limit charging. Can you add it to BatFi?

rurza, 2024-09-24:
Please update the app to the latest version. Everything is explained in the changelog.

closed 2024-09-24 by rurza

Neifab, 2025-05-15:
Hello, i'm on the last Sonoma 14.7.6 and the 2 options are not working. I am ending with a battery fully charge. Thanks