rurza/BatFi #94: Can I set the BatFi menu bar icon to the BatFi logo?

MasterOfMouse, 2025-01-16:
I want to keep using the macOS's own battery icon in the Menu bar, I don't want 2 instances of the battery icon.

rurza, 2025-01-24:
Hey!
You can hide it if you don't need it.

MasterOfMouse, 2025-03-27:
[image]

Then I wouldn't be able to see all this gorgeous data.

rurza, 2025-04-03:
Okay, now I get it :) I'm okay with it. Do you folks have any suggestion about the icon itself?

MasterOfMouse, 2025-04-10:
[image]

Rudimentary render, just made the BatFi logo b&w and dropped it in.

Making it match the rest of the icons should do the trick.

rurza, 2025-04-11:
WDYT?

https://github.com/user-attachments/assets/347ea254-bd12-45bd-a569-987976d089ef

rurza, 2025-04-11:
The icon is not final :) I have nicer design

[image]

MasterOfMouse, 2025-04-16:
> The icon on the video is not final :) I have nicer design
> 
> [image]

Love the little ears, it looks like a cute rabbit!