rurza/BatFi #100: Sometimes there pops up an "I need juice'-notification despite being well over 50% battery.

qns7, 2025-02-13:
... and I didn't even have notifications set in the settings. 

Best regards
Q

closed 2025-02-20 by rurza

qns7, 2025-02-23:
[image]

It's still there...

qns7, 2025-02-26:
[image]

Why got this issue closed? It still hapenns and I cannot figure out why.

qns7, 2025-03-10:
@rurza It's still there:

I reinstalled BatFi now after removing it "completely" with AppCleaner and (from the beginning) turned all notifications off inside the apps settings.

Perhaps it has something to do with me using yabai and/or having the menubar hidden?

rurza, 2025-03-10:
@qns7 Hi! Even on the version `2.3.2`?

qns7, 2025-03-11:
> [@qns7](https://github.com/qns7) Hi! Even on the version `2.3.2`?

Thanks for chiming in again ♥️
I ride 2.3.2 since it was released, but still had the issue.
Since my last post it didn't occur yet, so maybe it was a problem with my machine specifically and is fixed now.
Perhaps because of some interference with yabai or something - as mentioned - I don't know.
I'll keep you posted.

PS: off-topic: This morning after waking up my MacBook from the night BatFi showed me "Wallpaper" in "Apps with high energy usage" with a value at around 7XX. Do you know what that means?