rurza/BatFi #109: Bug Report: High Energy Impact, RAM & Port Usage Increase in BatFi 2.3.2 on MacOS 15.3.1 (M4 Pro)

NekoSosu, 2025-03-13:
BatFi Version: 2.3.2
Operating System: MacOS 15.3.1
Device: Macbook M4 Pro

Issue Report:

I have observed that the energy impact occasionally spikes up to 600.

Additionally, I've noticed a continuous increase in RAM usage. After leaving my Macbook running overnight, the RAM usage increased from 50MB to 200MB.  The number of ports also increased significantly, from 700 to 50,000.

[image]

[image]

[image]

rurza, 2025-03-13:
Ouch!
Thanks for the email and report!

cyanghxst, 2025-07-04:
I'm experiencing the same issue, and I found it kinda funny that BatFi reports itself under Apps with High Energy Usage.

- **BatFi**: 2.4.2
- **MacOS**: Sequoia 15.5

[image]

Terabyte1385, 2025-11-30:
Leaving my experience here too (M2 MacBook Air on macOS 26.1)

121,01 12-hr power
[image]

~300MB RAM and almost 100k ports open
[image]

I will be honest, I'm not entirely sure what these values mean/what they're supposed to be for BatFi, but it doesn't look normal.