rurza/BatFi #117: Stuck on Initializing macOS Tahoe 26.0 Beta (25A5279m)

plyght, 2025-06-10:


plyght, 2025-06-10:
let me know if any more context is needed

ongkiii, 2025-06-10:
[image]

most of the functionality is broken like battery limiter, menu bar icon, etc.

Final-Hawk, 2025-06-10:
I installed macOS Tahoe on a different partition and I am now having this issue on Sequoia, even after deleting the partition. I have tried reinstalling many times, safe mode, and permission reset commands.

plyght, 2025-06-10:
@Final-Hawk wow! it's widespread? super odd...

slrgt, 2025-06-21:
> Turn on Check for updates in Batfi and make sure beta updates are on it should Check for updates regularly and you will see a pop up on your screen telling you the version and whats included in that version. You also get the option to update now or later
> Could you please let me know if there have been any updates or an estimated timeline for compatibility?
> Thanks in advance for your time and help.
> Yes there has the ***@***.***(https://github.com/rurza) is planning on releasing the beta closer to when the Public stable version of MacOS Tahoe. From that I assume Last week of August or First week of September. Maybe even the day the public version is released. There is not a known date yet but The Dev is working hard to get everything working and setup however it cannot be promised that everything will work properly.
> Thanks for the update! That makes sense — option 2 sounds good then. I’ll plan around a release sometime in late August or early September. Really appreciate the effort ***@***.***(https://github.com/rurza) is putting in to make sure everything works smoothly. Looking forward to testing the beta when it’s ready!

Would be nice to have an alpha version with just the battery pausing working before then if possible

NasifJK, 2025-06-22:
> Would be nice to have an alpha version with just the battery pausing working before then if possible
> […](#)

It would but you need to remember that the app is probably really buggy right now so maybe when the public beta comes out that would be a good idea to release a alpha version

NasifJK, 2025-07-12:
> Hi, are there any updates on when we can expect a compatible version for macOS 26? When the public version is out? Thank you very much.

The dev plans to release a beta before the public version comes out in September
around the end of august or beginning of September

ladislavsulc, 2025-07-12:
Excellent, thank you for the update.

thanhhoangngoc, 2025-07-23:
> > > Okay, folks 🙂 I have some good news! I was able to disable charging on the latest firmware, but I’ll need at least a month to test it properly. I’m not saying it will have full feature parity with the current version, but of course, I’ll do my best.
> > > Here’s the plan:
> > > I’ll release a beta version 3.0 of the app closer to the public beta release of macOS. This version will be designed specifically for macOS 26 and will require it to run. It’s a good opportunity to refactor some code and make a few UI changes.
> > > The current (2.x) version of the app won’t support macOS 26. This means users who installed macOS 26 on a separate volume won’t be able to use BatFi on their current OS until they return to macOS 26. To get the app working again, you'll need to restore the firmware: https://support.apple.com/en-us/108900
> > > Friendly reminder: Please, never ever install macOS betas—especially major ones, and especially beta 1—unless you absolutely need to. These early versions are intended for Apple to collect initial feedback. No experienced developer working on a complex app will release an update targeting a beta, because Apple often changes behavior or fixes things later.
> > 
> > So now it works on latest macOS26 beta, thank you so much. But need to fix blank icon 💪[image]
> 
> Wait where did you get 3.0 from? 

From Batffi itself in settings 😃

DylanD0g, 2025-08-22:
Doesn't wark for my laptop on last beta 26.

thanhhoangngoc, 2025-08-22:
> Doesn't wark for my laptop on last beta 26.

Latest BatFi beta working fine on my latest beta 26[image]

DylanD0g, 2025-08-22:
> > Doesn't wark for my laptop on last beta 26.
> 
> Latest BatFi beta working fine on my latest beta 26[image]

The App is working fine but it doesn't limit power charging.

404mat, 2025-09-16:
Just wanted to add to this. After installing macOS 26, even restarting multiple times did not resolve the "initial state" issue. The good old `uninstall -> restart -> reinstall` did the trick. I think it is something with the helper because as soon as I clicked on "Install Helper" in the onboarding, it worked instantly.