rurza/BatFi #119: MacOS Sequoia 15.5/MBA M4 – Abnormal Wake & Startup Chime During Sleep

BAI-94, 2025-06-18:
**Title: macOS Sequoia 15.5/MBA M4 – Abnormal Wake & Startup Chime During Sleep**

**Mac Model:** MacBook Air M4
**macOS Version:** macOS Sequoia 15.5
**BatFi Version:** 2.4.2

**Issue Description:**
With BatFi set to a charge limit of 55%, the system frequently and periodically experiences abnormal wake events during sleep. These wake events are accompanied by the startup chime (despite "Play sound on startup" being disabled), and the screen briefly flashes a battery icon before the Mac returns to sleep.

**Steps to Reproduce:**

1. Set BatFi charge limit to 55% and allow battery to charge to exactly 55%.
2. Put Mac to sleep while connected only to the official MagSafe 3 adapter, with all other peripherals disconnected.

**Expected Behavior:**
The Mac should remain in quiet sleep. Any wake events should be silent “DarkWake” events for regular maintenance (e.g., RTC checks).

**Actual Behavior:**
The Mac frequently wakes from sleep with a startup chime. At the moment of wake, the screen sometimes briefly shows a low battery icon, then turns off again.

**pmset -g log:**
_Wakes typically triggered by `due to smc.70070000 USB-C_plug` events, eventually logged as `DarkWake to FullWake ... due to Notification`. No kernel panics or crash reports generated._

```
2025-06-18 04:41:02 +0800 Sleep                Entering Sleep state due to 'Maintenance Sleep':TCPKeepAlive=active Using AC (Charge:55%) 461 secs  

2025-06-18 04:48:43 +0800 DarkWake            DarkWake from Deep Idle [CDNPB] : due to smc.70070000 USB-C_plug SMC.OutboxNotEmpty nub-spmi-a0.0x3F/ Using AC (Charge:55%) 2 secs    

2025-06-18 04:48:45 +0800 Wake                DarkWake to FullWake from Deep Idle [CDNVAPB] : due to Notification Using AC (Charge:55%) 5 secs    

2025-06-18 04:48:50 +0800 Sleep               Entering DarkWake state due to 'Notification Wake Back to Sleep':TCPKeepAlive=active Using AC (Charge:55%)            

2025-06-18 04:49:30 +0800 Sleep               Entering Sleep state due to 'Notification Wake Back to Sleep':TCPKeepAlive=active Using AC (Charge:55%) 1 secs    

2025-06-18 04:49:31 +0800 DarkWake            DarkWake from Deep Idle [CDNPB] : due to smc.70070000 USB-C_plug SMC.OutboxNotEmpty nub-spmi-a0.0x3F/Notification Using AC (Charge:55%) 45 secs    

2025-06-18 04:50:16 +0800 Sleep               Entering Sleep state due to 'Notification Wake Back to Sleep':TCPKeepAlive=active Using AC (Charge:55%) 133 secs  

2025-06-18 04:52:29 +0800 DarkWake            DarkWake from Deep Idle [CDNPB] : due to smc.70070000 USB-C_plug SMC.OutboxNotEmpty nub-spmi-a0.0x3F/ Using AC (Charge:55%) 2 secs    

2025-06-18 04:52:31 +0800 Wake                DarkWake to FullWake from Deep Idle [CDNVAPB] : due to Notification Using AC (Charge:55%) 5 secs    

2025-06-18 04:52:36 +0800 Sleep               Entering DarkWake state due to 'Notification Wake Back to Sleep':TCPKeepAlive=active Using AC (Charge:55%)            

2025-06-18 04:53:14 +0800 Sleep               Entering Sleep state due to 'Notification Wake Back to Sleep':TCPKeepAlive=active Using AC (Charge:55%) 394 secs  

2025-06-18 04:59:48 +0800 DarkWake            DarkWake from Deep Idle [CDNPB] : due to smc.70070000 USB-C_plug SMC.OutboxNotEmpty nub-spmi-a0.0x3F/ Using AC (Charge:55%) 2 secs    

2025-06-18 04:59:50 +0800 Wake                DarkWake to FullWake from Deep Idle [CDNVAPB] : due to Notification Using AC (Charge:55%) 4 secs    
```

**Additional Notes:**
When testing without BatFi (i.e., just charging with MagSafe or with no charging), no abnormal wake behavior occurs.

**Conclusion:**
I suspect the root cause is that the system misinterprets BatFi-triggered charge state changes as physical power connect/disconnect events. This leads to an unnecessary and disruptive wake sequence.

I've spent several days investigating and isolating this issue. Would appreciate any possibility of optimization.

rurza, 2025-06-18:
Thank you for the detailed report :) I will take a look

BAI-94, 2025-06-20:
> Thank you for the detailed report :) I will take a look

Thank you for your attention to this issue. If there are any conclusions or plans for an optimized version, I would appreciate it if you could update me here.

closed 2025-07-05 by blakejett